ethercat-rs/ethercat #16: Version of IgH ethercat code

elias10101, 2020-09-04:
Nice work! Which version of the IgH Ethercat master code are you using for this wrapper?

birkenfeld, 2020-09-04:
I was working with the `default` branch of the hg repository.  Now I checked again and saw that apparently the IgH people are working exclusively on the `stable-1.5` branch...

But I think it shouldn't matter too much; it looks the ioctl interface we use between userspace and kernelspace hasn't changed except for a shift in the ioctl numbers. You just need to generate the Rust bindings with the version you're also building the kernel module from.

flosse, 2020-09-04:
Unfortunately there are differences... even between `v1.5.2-sncn-7` and `v1.5.2-sncn-11` (look at `master/ioctl.h`).
See #17  (PR to update to `v1.5.2-sncn-11`).

birkenfeld, 2020-09-04:
Ah damn, we aren't pulling the ioctl numbers from the header file... should we?

birkenfeld, 2020-09-04:
In fact, it should be quite straightforward to autogenerate the `ioctl!` definitions from the `#define`s in `ioctl.h`.

birkenfeld, 2020-09-04:
In any case, we should definitely point to the "canonical" branch we test with in the README.

flosse, 2020-09-09:
Done with #22

closed 2020-09-09 by flosse

flosse, 2020-09-09:
... and 9f24c1b6bd44c2c8ee2a3c942876351febfb80f2