ethercat-rs/ethercat #37: Error compiling ethercat-sys

17steen, 2023-02-16:
using latest stable revision of the recommended ethercat repo, all works when using the c library, getting this error while compiling:
<pre><font color="#00AA00"><b>   Compiling</b></font> ethercat-sys v0.2.0
<font color="#AA0000"><b>error</b></font><b>:</b> failed to run custom build command for `ethercat-sys v0.2.0`

Caused by:
  process didn&apos;t exit successfully: `/home/student/Sources/rust/ethercat-connection-rs/target/debug/build/ethercat-sys-3ae150160a12e9a6/build-script-build` (exit status: 101)
  --- stderr
  thread &apos;main&apos; panicked at &apos;&quot;ec_ioctl_slave_t_struct_(unnamed_at_/home/student/Downloads/ethercat/master/ioctl_h_228_5)&quot; is not a valid Ident&apos;, /home/student/.cargo/registry/src/github.com-1ecc6299db9ec823/proc-macro2-1.0.51/src/fallback.rs:752:9
  note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
</pre>

using Ubuntu 22.04.1 LTS
kernel: 5.19.0-32-generic

17steen, 2023-02-16:
these are my configure flags: `./configure --enable-generic --disable-8139too --disable-e100 --disable-e1000 --disable-e1000e --disable-igb --disable-r8169 --disable-ccat --sysconfdir=/etc`

birkenfeld, 2023-02-17:
Can you share more details about your checkout of the master repo? You should switch to the `stable-1.5` branch.

birkenfeld, 2023-02-17:
I just realized this is not documented (except on IgH's site) and pushed an addition to README.

17steen, 2023-02-17:
Sorry forgot to mention that, but I checked the stable branch out aswell

<pre>* <font color="#00AA00">stable-1.5</font>
</pre>

birkenfeld, 2023-02-17:
Hm. Can you set the `RUST_BACKTRACE=1` env var and post the traceback?

birkenfeld, 2023-02-17:
I cannot reproduce with a fresh Ubuntu 22.04. Below is the Dockerfile I used...

```Dockerfile
FROM ubuntu:22.04

RUN apt-get update \
 && apt-get install -y autoconf libtool pkg-config make systemd git cargo linux-headers-generic libclang-dev

RUN git clone https://gitlab.com/etherlab.org/ethercat
RUN cd ethercat && git checkout stable-1.5 && ./bootstrap \
 && ./configure --disable-8139too --with-linux-dir=/usr/src/linux-headers-*-generic

RUN git clone https://github.com/ethercat-rs/ethercat ethercat-rs
RUN cd ethercat-rs && ETHERCAT_PATH=/ethercat cargo build
```

17steen, 2023-02-17:
maybe i should start with a clean repo, i had configured with other flags before

17steen, 2023-02-17:
<pre><font color="#00AFFF">RUST_BACKTRACE</font><font color="#00A6B2">=</font><font color="#00AFFF">1</font> <font color="#005FD7">cargo</font> <font color="#00AFFF">build</font>
<font color="#00AA00"><b>   Compiling</b></font> ethercat-sys v0.2.0
<font color="#AA0000"><b>error</b></font><b>:</b> failed to run custom build command for `ethercat-sys v0.2.0`
note: To improve backtraces for build dependencies, set the CARGO_PROFILE_DEV_BUILD_OVERRIDE_DEBUG=true environment variable to enable debug information generation.

Caused by:
  process didn&apos;t exit successfully: `/home/student/Sources/rust/ethercat-connection-rs/target/debug/build/ethercat-sys-3ae150160a12e9a6/build-script-build` (exit status: 101)
  --- stderr
  error: header &apos;/home/student/Download/ethercat//lib/ioctl.h&apos; does not exist.
  thread &apos;main&apos; panicked at &apos;Unable to generate bindings: ()&apos;, /home/student/.cargo/registry/src/github.com-1ecc6299db9ec823/ethercat-sys-0.2.0/build.rs:26:14
  stack backtrace:
     0: rust_begin_unwind
               at /rustc/5b8f284536d00ba649ca968584bedab4820d8527/library/std/src/panicking.rs:575:5
     1: core::panicking::panic_fmt
               at /rustc/5b8f284536d00ba649ca968584bedab4820d8527/library/core/src/panicking.rs:64:14
     2: core::result::unwrap_failed
               at /rustc/5b8f284536d00ba649ca968584bedab4820d8527/library/core/src/result.rs:1790:5
     3: core::result::Result&lt;T,E&gt;::expect
     4: build_script_build::main
     5: core::ops::function::FnOnce::call_once
  note: Some details are omitted, run with `RUST_BACKTRACE=full` for a verbose backtrace.
</pre>

17steen, 2023-02-17:
```toml
[package]
name = "ethercat-connection-rs"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
ethercat = "0.2.3"
```

this is my Cargo.toml

birkenfeld, 2023-02-17:
bindgen seems to get stuck at the anonymous sub-struct in  `master/ioctl.h` line 228. I can only imagine an incompatibility with clang (bindgen uses `libclang` to parse the C code). Which clang/libclang do you have installed? Can you try with the latest one, like is installed in my Docker example?

17steen, 2023-02-18:
I am using clang-17 from LLVM's ppa's this might be it, I do not exactly know how I can instruct the build process to use the ones provided by the ubuntu repositories instead

17steen, 2023-02-18:
maybe i can LD_PRELOAD an older version of libclang

17steen, 2023-02-18:
managed to build it using a fresh container as well.
Once the sys package is built i can go forth outside the container, I assume, so this resolves the issue.

The unstable version of clang on my system must be the cause of the problem.
Thanks for helping me find that out.

closed 2023-02-18 by 17steen

birkenfeld, 2023-02-18:
Thanks for confirming! We also have the `pregenerated-bindings` feature which removes the need for the C checkout completely, however it is tied to a specific (and now outdated) revision.

Since IgH don't do release versions anymore, it's kind of hard to decide when to update them :(