ethercat-rs/ethercat #39: error[E0308] while building

spiderdab, 2023-02-23:
Hi, I was trying to build this project but had this error:
`user@raspberrypi:~/rust_ethercat/ethercat $ cargo build
   Compiling ethercat v0.2.3 (/home/dab/rust_ethercat/ethercat)
error[E0308]: mismatched types
  --> src/convert.rs:64:29
   |
64 |     unsafe { CStr::from_ptr(data).to_string_lossy().into_owned() }
   |              -------------- ^^^^ expected `u8`, found `i8`
   |              |
   |              arguments to this function are incorrect
   |
   = note: expected raw pointer `*const u8`
              found raw pointer `*const i8`
note: associated function defined here
  --> /rustc/d5a82bbd26e1ad8b7401f6a718a9c57c96905483/library/core/src/ffi/c_str.rs:256:25`

I'm on a raspberry pi 400 with latest 64bit raspberry pi OS.
Ethercat from igh is installed and working (and master is started).
Rust and Cargo version are 1.67.1

is it a known issue?

birkenfeld, 2023-02-23:
No, but confusion with the char definition on arm are common. I'll fix it soon.

birkenfeld, 2023-02-24:
Can you verify that #40 fixes compilation?

spiderdab, 2023-02-24:
Thank you, I can confirm it fixed the issue.

closed 2023-02-24 by spiderdab

birkenfeld, 2023-02-24:
0.2.4 is released with the fix.

commit 45f6d49d (2023-02-24): fix missing casts from i8 to c_char

Fixes #39
PR #40 (2023-02-24): fix missing casts from i8 to c_char
Fixes #39
commit d59108e1 (2023-02-24): fix missing casts from i8 to c_char

Fixes #39