ethercat-rs/ethercat #41: Info needed for rust novice.

spiderdab, 2023-02-24:
Hi, I'm a rust novice, but used ethercat c lib for the last 2 years.
I wanted to try to go the rust way, that's why I'm trying this repo, but I can't go on with the examples, possibly for my limited knowledge on how to use rust libraries. 

I succesfully built ethercat from Igh on a raspberry pi 400 with latest 64bit raspberry pi OS. Master runs ok and a slave is correctly found.
Than, thanks to #40 , I could build ethercat-rs doing:
`
export ETHERCAT_PATH=~/ethercat
cargo build
`

I see that examples are not built, so I've tried:

`
$ rustc examples/info.rs
$ rustc examples/info.rs -L ./
`
also tried adding 
`
extern crate ethercat;
`
to the beginning of info.rs, but in every case the result is that ethercat is missing.

What should I do to run an example?
Would it be possible to write in the documentation a simple list of commands to run an example?

spiderdab, 2023-02-24:
Thanks, I can confirm it works with this result:
`
dab@raspberrypi:~/rust_ethercat/ethercat $ cargo run --example info
    Finished dev [unoptimized + debuginfo] target(s) in 0.11s
     Running `target/debug/examples/info`
EtherCAT Master: Ok(
    MasterInfo {
        slave_count: 1,
        link_up: true,
        scan_busy: false,
        app_time: 0,
    },
)
`

I have to say I missed (as a rust novice) that cargo has a special flag to run examples...

closed 2023-02-24 by spiderdab

birkenfeld, 2023-02-24:
No problem, there is a lot in there :)