ethercat-rs/ethercat #44: How to configure `Ethercat master` in `build.rs`?

rise0chen, 2023-05-11:
I want to configure `Ethercat master`  in `build.rs` of my project.
```rs
fn main() -> Result<(), Box<dyn std::error::Error>> {
    use std::process::Command;
    use std::{env, fs};
    let dir = fs::canonicalize("./vendor/etherlab")?;
    env::set_var("ETHERCAT_PATH", &dir);
    env::set_current_dir(&dir)?;
    Command::new(format!("{}/bootstrap", dir.display())).current_dir(&dir).status().unwrap();
    Command::new(format!("{}/configure", dir.display()))
        .current_dir(&dir)
        .arg("--disable-8139too")
        .arg("--enable-igb")
        .arg("--enable-cycles")
        .arg("--enable-rtdm")
        .arg("--enable-hrtimer")
        .arg("--with-rtai-dir=/usr/realtime")
        .arg("--with-linux-dir=/usr/src/linux")
        .status()
        .unwrap();
    Ok(())
}
```

It should run `build.rs` of my project first, and then run `build.rs` of `ethercat-sys`.
But in fact, `build.rs` of `ethercat-sys` is first.
Do you have any suggestions?

rise0chen, 2023-05-12:
No, I   not want to use `pregenerated-bindings`.

I have success to run this project by manual configure.
I run `bootstrap` , `configure` and set `ETHERCAT_PATH` env, then run `cargo run`, it can work.
I just wat to configure it by `build.rs`.

birkenfeld, 2023-05-12:
As far as I know, there is no way to have `build.rs` or anything else run before the dependencies `build.rs`.

rise0chen, 2023-05-12:
Can I call `ethercat-sys/build.rs` in my `build.rs`?

birkenfeld, 2023-05-14:
That wouldn't help since at that time `ethercat-sys/build.rs` already has run.

closed 2024-08-10 by birkenfeld