ethercat-rs/ethercat #45: Latest version not compiling

jubeor-gp, 2023-07-10:
First time use with the next setup:
- OS: Ubuntu Server 22.04 minimised
- Kernel: 5.15.0-76-generic
- ethercat-igh: ethercat slaves shows slaves for my ethercat NIC
- Cargo Version: 1.70.0 (ec8a8a0ca 2023-04-25)

## Steps

Just cloned latest version from this repo (`git clone https://github.com/ethercat-rs/ethercat.git`)
exporting ETHERCAT_PATH so ethercat-sys can find it:
```#bash
$> ETHERCAT_PATH=/home/user/ethercat-1.5.0/ethercat/
$> export ETHERCAT_PATH
```

- ethercat-sys is build successfully but ethercat fails:

```
$> cargo build
   Compiling ethercat v0.3.0 (/home/gpg/ethercat-test/ethercat)
error[E0308]: mismatched types
  --> src/convert.rs:27:8
   |
27 |     Ok(foe_name)
   |     -- ^^^^^^^^ expected an array with a fixed size of 255 elements, found one with 32 elements
   |     |
   |     arguments to this enum variant are incorrect
   |
help: the type constructed contains `[i8; 32]` due to the type of the argument passed
  --> src/convert.rs:27:5
   |
27 |     Ok(foe_name)
   |     ^^^--------^
   |        |
   |        this argument influences the type of `Ok`
note: tuple variant defined here
  --> /rustc/90c541806f23a127002de5b4038be731ba1458ca/library/core/src/result.rs:507:5

For more information about this error, try `rustc --explain E0308`.
error: could not compile `ethercat` (lib) due to previous error

```

## Workaround

Edit convert.rs so the function `string_to_foe_name` admits and returns a 255 array.

Now it compiles.  All examples run:

```
$> cargo run --example
error: "--example" takes one argument.
Available examples:
    cyclic-data
    foe-read
    foe-write
    info
    sdo

$> cargo run --example cyclic-data
    Finished dev [unoptimized + debuginfo] target(s) in 0.04s
     Running `target/debug/examples/cyclic-data`
usage: ethercat ESI-FILE
$> cargo run --example foe-read
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/examples/foe-read`
Usage: foe-read <slave-position> <foe-name>
Error: Custom { kind: Other, error: "Not enough arguments" }
$> cargo run --example foe-write
   Compiling ethercat v0.3.0 (/home/gpg/ethercat-test/ethercat)
   Finished dev [unoptimized + debuginfo] target(s) in 0.39s
     Running `target/debug/examples/foe-write`
Usage: foe-read <slave-position> <foe-name> <file>
Error: Custom { kind: Other, error: "Not enough arguments" }
$> cargo run --example foe-write
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/examples/foe-write`
Usage: foe-read <slave-position> <foe-name> <file>
Error: Custom { kind: Other, error: "Not enough arguments" }
$> cargo run --example info
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/examples/info`
EtherCAT Master: Ok(
    MasterInfo {
        slave_count: 8,
        link_up: true,
        scan_busy: false,
        app_time: 0,
    },
)
$> cargo run --example sdo
    Finished dev [unoptimized + debuginfo] target(s) in 0.02s
     Running `target/debug/examples/sdo`
Could not find any SDOs
```

I hope this helps.

birkenfeld, 2023-07-10:
Please check out the `stable-1.5` branch in the IgH repository. This library works against that branch, not `master`.

closed 2023-07-10 by birkenfeld

jubeormk1, 2023-07-12:
Hi again,

I get your point checking out the `stable-1.5` branch of IgH repository as it is indicated in the readme file. This repo is not responsible for changes in IgH repositories.

The modification I did to this repo code to make it run was to change the hardcoded size of an array and some assertions about it. the error thrown by cargo is `E0308` mismatching types error. I might be wrong but the issue that I am facing has to do with rust code.

I am aware that you are an expert in rust while I am just starting. Maybe I mislead you with the title of the issue?

jubeor-gp, 2023-07-12:
Hi again,

I get your point checking out the stable-1.5 branch of IgH repository as it is indicated in the readme file. This repo is not responsible for changes in IgH repositories.

The modification I did to this repo code to make it run was to change the hardcoded size of an array and some assertions about it. the error thrown by cargo is E0308 mismatching types error. I might be wrong but the issue that I am facing has to do with rust code.

I am aware that you are an expert in rust while I am just starting. Maybe I mislead you with the title of the issue? In any event, thanks for this repository.

birkenfeld, 2023-07-12:
I'm sorry, I don't fully understand your comment. After you switch the IgH code to the `stable-1.5` branch, this library should compile fine. One of the changes between `stable-1.5` and `master` is that they changed the size of the array in question from 32 to 255 items.

You may have to `cargo clean` before rebuilding though, I don't think the build system is clever enough to realize that the C header has changed.