ethercat-rs/ethercat #47: cyclic_data example no longer compiles

askuric, 2024-04-25:
Hi guys, 

I'm having trouble compiling the cycling_data example. 
Here is the message:
```
error[E0609]: no field `index` on type `&Pdo`
   --> src/bin/cyclic_data1.rs:125:39
    |
125 |                 idx: PdoIdx::from(pdo.index),
    |                                       ^^^^^ unknown field
    |
    = note: available fields are: `sm`, `fixed`, `mandatory`, `idx`, `name`, `entries`

error[E0609]: no field `index` on type `&PdoEntry`
   --> src/bin/cyclic_data1.rs:132:46
    |
132 | ...                   idx: Idx::from(e.index),
    |                                        ^^^^^ unknown field
    |
    = note: available fields are: `entry_idx`, `bit_len`, `name`, `data_type`

error[E0609]: no field `sub_index` on type `&PdoEntry`
   --> src/bin/cyclic_data1.rs:133:53
    |
133 | ...                   sub_idx: SubIdx::from(e.sub_index.unwrap_or(1) as u8),
    |                                               ^^^^^^^^^ unknown field
```

Could this be a problem of my configuration or is this due to some API change?
Thanks

birkenfeld, 2024-08-10:
I can't reproduce, at least with current master branch. Please let me know if you're still having problems.

askuric, 2024-08-10:
The issue did not go away

birkenfeld, 2024-08-10:
Ok, reopening to clarify this. The `examples/cyclic-data.rs` currently on master definitely compiles with the lib from master.

Which code did you take, and with which version of the lib are you using it?

askuric, 2024-08-11:
Hi @birkenfeld,

Thanks for the message!

I've tested some more and it seems that it is the version of the `ethercat-esi` that causes these problems. 
Basically the examples are using the older version `0.1` and in my case the version that was downloaded by default was the latest `0.2`. And the new version is braking the API and the example does no longer compile. 

So you are absolutely right, if you build the example by downloading the master branch and compiling it it will work.

In case you need to create a new cargo project and if you copy the `cyclic-data` example inside and not specify explicitly the `ethercat-esi` version to be `0.1`, the code will not compile. 

Here is the `Cargo.toml` that compiles and runs the `cyclic-data` example for those interested :D

```toml
[package]
name = "cyclic_data_test"
version = "0.1.0"
edition = "2021"

[dependencies]
ethercat = "0.3.1" 
log = "0.4"
ethercat-esi = "0.1.0" # important
env_logger = "0.8"
```

birkenfeld, 2024-08-24:
Gotcha! I've updated the `ethercat-esi` dependency to 0.2 in the example now.

commit fa554eeb (2024-08-24): bump to ethercat-esi 0.2 for cyclic-data info example

fixes #47
closed 2024-08-24 by birkenfeld